pimusicbox/pimusicbox #480: Spotify connect needs an updated librespot - Thanks

nemul, 2018-05-23:


MartyTremblay, 2018-05-29:
Thanks @kingosticks 

I just tried, and unless something has changed since, it doesn't work for me. I'll open a ticket with librespot if it makes sense. I now get the following:

INFO:librespot: librespot aeb205d (2018-03-22). Built on 2018-05-21. Build ID: gB1rceMk
INFO:librespot_core::session: Connecting to AP "guc3-accesspoint-b-m84k.ap.spotify.com:4070"
INFO:librespot_core::session: Authenticated as "___________" !
INFO:librespot_core::session: Country: "CA"
INFO:librespot_playback::audio_backend::alsa: Using alsa sink
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: WireError(InvalidEnumValue(14))', /checkout/src/libcore/result.rs:916:5
stack backtrace:
   0:   0x939d17 - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::h280c621ec0336843
                       at /checkout/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1:   0x943f73 - std::sys_common::backtrace::print::h5bb167d48668dac9
                       at /checkout/src/libstd/sys_common/backtrace.rs:68
                       at /checkout/src/libstd/sys_common/backtrace.rs:57
   2:   0x93259f - std::panicking::default_hook::{{closure}}::hccf3b2e1cc48596f
                       at /checkout/src/libstd/panicking.rs:381
   3:   0x93207b - std::panicking::default_hook::h338b1189515ccc69
                       at /checkout/src/libstd/panicking.rs:397
   4:   0x51f763 - librespot::main::{{closure}}::h36f160aa49878201
   5:   0x932a3f - std::panicking::rust_panic_with_hook::h5d950fb4a856fba5
                       at /checkout/src/libstd/panicking.rs:578
   6:   0x9328ef - std::panicking::begin_panic::h7a5157aecf168062
                       at /checkout/src/libstd/panicking.rs:538
   7:   0x93281b - std::panicking::begin_panic_fmt::hc44606ee40e9a98e
                       at /checkout/src/libstd/panicking.rs:522
   8:   0x9327c3 - rust_begin_unwind
                       at /checkout/src/libstd/panicking.rs:498
   9:   0x99b87b - core::panicking::panic_fmt::h68e7715928be641f
                       at /checkout/src/libcore/panicking.rs:71
  10:   0x5b2017 - core::result::unwrap_failed::hc8e405a7d6f5e105
  11:   0x58a9a3 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &'a mut F>::call_once::h550fb1c59c949344
  12:   0x5a985b - <futures::stream::map::Map<S, F> as futures::stream::Stream>::poll::h6f4042a85363f6fe
  13:   0x5944ff - <librespot_connect::spirc::SpircTask as futures::future::Future>::poll::h2ae68a0a355e9093
  14:   0x51d107 - <librespot::Main as futures::future::Future>::poll::ha34952329266c131
  15:   0x51568f - futures::task_impl::std::set::haa73a9ee0a468b94
  16:   0x52e1a3 - tokio_core::reactor::Core::run::h8b40cb76d6ebd0af
  17:   0x51ec83 - librespot::main::hf84f741837bd497f
  18:   0x52ec2b - std::rt::lang_start::{{closure}}::h369662dc239e3883
  19:   0x9326f3 - std::panicking::try::do_call::h9cb586bd57bb9d3d
                       at /checkout/src/libstd/rt.rs:59
                       at /checkout/src/libstd/panicking.rs:480
  20:   0x95930b - __rust_maybe_catch_panic
                       at /checkout/src/libpanic_unwind/lib.rs:101
  21:   0x9399a3 - std::rt::lang_start_internal::h104139376fdb9ead
                       at /checkout/src/libstd/panicking.rs:459
                       at /checkout/src/libstd/panic.rs:365
                       at /checkout/src/libstd/rt.rs:58
  22:   0x51f79b - main
  23: 0xb6dbe81b - __libc_start_main

kingosticks, 2018-05-29:
Hmm, that's weird. I don't think there's any need to open an issue with librespot just yet, they did fix exactly this. I need to check the build I uploaded../

JayGatsby7, 2018-05-29:
Confirming that the original fix didn’t work on my end either. Thanks!

kingosticks, 2018-05-29:
New one is up:
```
service librespot stop
cd /opt/librespot
wget https://github.com/pimusicbox/librespot/releases/download/v20180529-1e69138/librespot-linux-armhf-raspberry_pi.zip
unzip librespot-linux-armhf-raspberry_pi.zip
rm librespot-linux-armhf-raspberry_pi.zip
service librespot start
```

I realise how I did this: forgot you have to also rebuild the docker container so that it picks up the new source code. Not fool-proof enough for me.

chimj, 2018-05-30:
Solved here as well, thanks.

nemul, 2018-05-30:
Works for me, thanks.

thescottyq, 2018-06-04:
Hi all, just ran this on SSH on our MusicBox and it still wont load playlists. Have I got to do anything else?

kingosticks, 2018-06-04:
This fixes the librespot issue which is the software for providing Spotify connect. The general Spotify support (accessed through the Webclients/mpd clients) is provided by mopidy-spotify which is still broken.

stevenmcastano, 2018-06-05:
Yup, this fixed it on mine as well... which is AWESOME! This is probably by far the single most used part of MusicBox for me! Thanks!

JorritBootsma, 2018-07-11:
@kingosticks somehow I get the message: 

'librespot: unrecognized service' 

both when I try 'service librespot stop' and 'service librespot start'. Did I install something incorrectly?

kingosticks, 2018-07-11:
Sounds like it but I don't know how you managed that. The service file lives in [/etc/init.d/librespot](https://github.com/pimusicbox/pimusicbox/blob/develop/filechanges/etc/init.d/librespot) but the upgrade commands I gave above do not touch that file.

JorritBootsma, 2018-07-11:
Hmm..  /etc/init.d/librespot does not exist

kingosticks, 2018-07-11:
librespot only provides Spotify Connect support. Everything else comes from mopidy-spotify so (perhaps surprisingly) they are entirely unrelated issues. If you want to start a more appropriate separate issue for spotify playlists on here you are welcome to.

The fix for latest mopidy-spotify is almost done. It may even get released into pimusicbox first due to the fact I won't bother fixing the tests when backporting it to pimusicbox's old version of mopidy-spotify...  But I don't have an ETA.

kingosticks, 2018-07-11:
@JorritBootsma Are you using the latest RC at https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC6 ? It doesn't sound like you are...

JorritBootsma, 2018-07-11:
I got that impression too so I formatted my SD-card and and made sure that I installed the right image so I should be now. However I am having some issues ssh-ing into my MusicBox now (I recognize having them before). Please hold.

JorritBootsma, 2018-07-11:
This clearly illustrates the problem:

root@MusicBox:~# service librespot      
Usage: /etc/init.d/librespot {start|stop|status|restart|force-reload|run}
root@MusicBox:~# service librespot status
[FAIL] librespot is not running ... failed!
root@MusicBox:~# service librespot start 
[ ok ] Starting Spotify Connect : librespot.
root@MusicBox:~# service librespot status
[FAIL] librespot is not running ... failed!
root@MusicBox:~# service librespot restart
[ ok ] Restarting Spotify Connect: librespot.
root@MusicBox:~# service librespot status
[FAIL] librespot is not running ... failed!
root@MusicBox:~# service librespot force-reload
[ ok ] Restarting Spotify Connect: librespot.
root@MusicBox:~# service librespot status
[FAIL] librespot is not running ... failed!

kingosticks, 2018-07-11:
Did you enter your Spotify credentials in the Spotify section of settings.ini? 

If you did enter your credentials properly, what does the log say in /var/log/librespot.log

JorritBootsma, 2018-07-11:
Yes I have, including Client ID and Client Secret.

This is the librespot.log file:

error: Required option 'name' missing
Usage: /opt/librespot/librespot [options]

Options:
    -c, --cache CACHE   Path to a directory where files will be cached.
        --disable-audio-cache 
                        Disable caching of the audio data.
    -n, --name NAME     Device name
        --device-type DEVICE_TYPE
                        Displayed device type
    -b, --bitrate BITRATE
                        Bitrate (96, 160 or 320). Defaults to 160
        --onevent PROGRAM
                        Run PROGRAM when playback is about to begin.
    -v, --verbose       Enable verbose output
    -u, --username USERNAME
                        Username to sign in with
    -p, --password PASSWORD
                        Password
        --proxy PROXY   HTTP proxy to use when connecting
        --disable-discovery 
                        Disable discovery mode
        --backend BACKEND
                        Audio backend to use. Use '?' to list options
        --device DEVICE Audio device to use. Use '?' to list options if using
                        portaudio
        --mixer MIXER   Mixer to use
        --initial-volume VOLUME
                        Initial volume in %, once connected (must be from 0 to
                        100)
        --zeroconf-port ZEROCONF_PORT
                        The port the internal server advertised over zeroconf
                        uses.
        --enable-volume-normalisation 
                        Play all tracks at the same volume
        --normalisation-pregain PREGAIN
                        Pregain (dB) applied by volume normalisation
        --linear-volume 
                        increase volume linear instead of logarithmic.

JorritBootsma, 2018-07-11:
************************
Initializing MusicBox...
************************
Setting sound configuration...
Probing sound devices:
* Found internal device: card0
* Found internal device: card0
Selected card=0  (i2s=  output=analog  usb=  intc=0)
Using audio card0 (analog)
MusicBox name is MusicBox
ifup: interface wlan0 already configured
Running /etc/init.d/networking restart is deprecated because it may not re-enable some interfaces ... (warning).
Reconfiguring network interfaces...Internet Systems Consortium DHCP Client 4.2.2
Copyright 2004-2011 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/b8:27:eb:ca:e7:b6
Sending on   LPF/wlan0/b8:27:eb:ca:e7:b6
Sending on   Socket/fallback
DHCPRELEASE on wlan0 to 192.168.2.254 port 67
Reloading /etc/samba/smb.conf: smbd only.
RTNETLINK answers: No such process
Internet Systems Consortium DHCP Client 4.2.2
Copyright 2004-2011 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/eth0/b8:27:eb:9f:b2:e3
Sending on   LPF/eth0/b8:27:eb:9f:b2:e3
Sending on   Socket/fallback
done.
Stopping Samba daemons: nmbd smbd.
Starting Samba daemons: nmbd smbd.
Starting Dropbear SSH server: dropbear.
Starting daemon monitor: monit.
Scanning music-files, please wait...
INFO     Starting Mopidy 1.1.2
INFO     Loading config from builtin defaults
INFO     Loading config from /etc/mopidy/mopidy.conf
INFO     Loading config from command line options
***********************************************************************************
My IP address is 192.168.2.31
Connect to me in your browser at http://MusicBox.local or http://192.168.2.31
***********************************************************************************

kingosticks, 2018-07-11:
And, you surely did this but I need to check you actually enabled the Spotify Connect service in settings.ini ? YOu do that by setting `enable_connect = true`....

JorritBootsma, 2018-07-11:
No.. I feel embarrassed. 

Did learn how to check log files, though... Sorry for wasting your time.

RubenRPINoob, 2018-08-26:
I am a little late to this tread, but I just installed pi MusicBox to my RPI zero w.

In my spotify app I can see the MusicBox between the devices to connect. When I try to connect it says "connecting.." and then after a while it stops and the MusicBox does't show up in the list anymore.

I tried the command lines above but when I type in the   wget "link"   error 404 NOT FOUND is stated.

Can I stream sporify music from my phone to MusicBox without singing in in the config file? Or do you have to sign in into the config file?

Sorry for my bad English it's not my first language.
best regards.

codethief, 2018-09-07:
@kingosticks Could I just build librespot myself (following their instructions) and use it as drop-in replacement for the version PiMusicBox provides? I'm asking because [there's been another issue](https://github.com/librespot-org/librespot/issues/223) which doesn't seem to be fixed yet in the version you uploaded.

kingosticks, 2018-09-07:
Yes you can do that. However, I'm pretty sure I picked up that fix in the build I already made available on May 29th.

codethief, 2018-09-07:
I assume you're talking about [this build](https://github.com/pimusicbox/pimusicbox/issues/480#issuecomment-392972884)? Maybe I made a mistake somewhere but I downloaded precisely this build and followed your instructions and the librespot log ended up saying "build on 2018-03-11". I'll double-check when I'm home again on Sunday.

codethief, 2018-09-09:
@kingosticks I checked again and you're right, I must have made some mistake along the way. It's working beautifully, now! Thanks again!

laurensV, 2018-10-24:
The new build fixes Spotify Connect for me (able to see MusicBox in Spotify and able to start playing music). However, it doesn't automatically stop whatever is playing on Modipy, so if you already have something playing on Modipy and start a song with Spotify Connect it plays the two simultaneously. Do I have to manually stop Modipy before I can use Spotify Connect or can this be done automatically? 

Would there be a way to better integrate Modipy and Spotify Connect? (ideally I would like to be able to also see and control the song I am playing through Spotify Connect in Modipy)

kingosticks, 2018-10-24:
Mopidy.

Librespot is configured to send a stop command to Mopidy when playback starts. I wonder why this is not working. If you enable the Mopidy debug log (in settings) can you check if Mopidy is receiving a stop command via its MPD interface?

Unfortunately integrating the two systems is not at all simple in their current states. I do agree it would be good to have.

tiramiseb, 2018-11-24:
@kingosticks : I've reinstalled piMusicBox twice recently, went on this thread twice and downloaded the wrong archive twice, while copy-pasting your first answer. :sweat_smile: 
Maybe you can edit your first answer and add something like "EDIT: wrong link, [see next answers](#issuecomment-392972884)".

raulgtrz23, 2019-07-16:
Just downloaded the latest release and unfortunately it is still not letting me switch to a new song and instead restarts the current song. What can I do?