pimusicbox/pimusicbox #483: missing kernel headers, wifi dongle

2000game, 2018-06-12:
I recently installed the pimusicbox software on my Pi1. In order to use the wifi I have the tplink wn-823n wifi dongle. My Kernel Version is 4.14.26+. When I check my USB devices via lsbusb I am getting this

> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 001 Device 004: ID 2357:0109  ** <- WiFi Dongle **
> Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
> Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub

The marked one is the wifi dongle. I already tried installing the drivers via the installation tutorial in the [drivers zip](https://static.tp-link.com/2018/201805/20180514/TP-Link_Driver_Linux_series8_beta.zip). Now I haven't tried modifying the kernel source because the dongle should work with my kernel. So I skipped that step. But I'm getting errors at the last make at 2.2. (Page 8 in the pdf). The make clean worked without errors.

> make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.26+/build M=/tmp/rtl8192EU_WiFi_linux_v5.2.19.1_25633.20171222_COEX20171113-0047  modules
> make[1]: *** /lib/modules/4.14.26+/build: No such file or directory.  Stop.
> Makefile:1828: recipe for target 'modules' failed
> make: *** [modules] Error 2

The Error of the make command

So it looks like I'm missing the kernel headers but I can't find out how to get them. I'm using the newest musicbox version (v0.7.0RC6)

kingosticks, 2018-06-12:
Did you try the usual `sudo apt-get install linux-headers-$(uname -r)`?

kingosticks, 2018-06-12:
Failing that, https://github.com/notro/rpi-source/wiki

2000game, 2018-06-12:
> sudo apt-get install linux-headers-$(uname -r)
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> E: Unable to locate package linux-headers-4.14.26
> E: Couldn't find any package by regex 'linux-headers-4.14.26'

Already tried that

2000game, 2018-06-12:
So I got it to work but still having problems loading the driver. 

After compiling the source I did this

> root@MusicBox:/tmp/rtl8192EU_WiFi_linux_v5.2.19.1_25633.20171222_COEX20171113-0047# sudo cp 8192eu.ko /lib/modules/4.14.26+/kernel/drivers/net/wireless/
> root@MusicBox:/tmp/rtl8192EU_WiFi_linux_v5.2.19.1_25633.20171222_COEX20171113-0047# sudo depmod -a
> root@MusicBox:/tmp/rtl8192EU_WiFi_linux_v5.2.19.1_25633.20171222_COEX20171113-0047# sudo modprobe 8192eu
> ERROR: could not insert '8192eu': Unknown symbol in module, or unknown parameter (see dmesg)

dmesg shows the the following

> [    0.000000] Booting Linux on physical CPU 0x0
> [    0.000000] Linux version 4.14.26+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1099 Wed Mar 14 14:35:31 GMT 2018
> [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
> [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
> [    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Rev 2
> [    0.000000] Memory policy: Data cache writeback
> [    0.000000] cma: Reserved 8 MiB at 0x1e400000
> [    0.000000] On node 0 totalpages: 126976
> [    0.000000] free_area_init_node: node 0, pgdat c099c2dc, node_mem_map ddfa4000
> [    0.000000]   Normal zone: 1116 pages used for memmap
> [    0.000000]   Normal zone: 0 pages reserved
> [    0.000000]   Normal zone: 126976 pages, LIFO batch:31
> [    0.000000] random: fast init done
> [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
> [    0.000000] pcpu-alloc: [0] 0
> [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
> [    0.000000] Kernel command line: bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.fiq_enable=1 dwc_otg.fiq_fsm_enable=1 dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
> [    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
> [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
> [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
> [    0.000000] Memory: 484128K/507904K available (6332K kernel code, 562K rwdata, 1964K rodata, 436K init, 677K bss, 15584K reserved, 8192K cma-reserved)
> [    0.000000] Virtual kernel memory layout:
> [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
> [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
> [    0.000000]     vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
> [    0.000000]     lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
> [    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
> [    0.000000]       .text : 0xc0008000 - 0xc0637268   (6333 kB)
> [    0.000000]       .init : 0xc08ad000 - 0xc091a000   ( 436 kB)
> [    0.000000]       .data : 0xc091a000 - 0xc09a6828   ( 563 kB)
> [    0.000000]        .bss : 0xc09ac174 - 0xc0a55634   ( 678 kB)
> [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
> [    0.000000] ftrace: allocating 23767 entries in 70 pages
> [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
> [    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
> [    0.000073] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
> [    0.000163] bcm2835: system timer (irq = 27)
> [    0.000720] Console: colour dummy device 80x30
> [    0.001339] console [tty1] enabled
> [    0.001392] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
> [    0.060341] pid_max: default: 32768 minimum: 301
> [    0.060866] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
> [    0.060922] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
> [    0.062232] Disabling memory control group subsystem
> [    0.062443] CPU: Testing write buffer coherency: ok
> [    0.063575] Setting up static identity map for 0x8200 - 0x8238
> [    0.064875] devtmpfs: initialized
> [    0.074736] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
> [    0.075200] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
> [    0.075274] futex hash table entries: 256 (order: -1, 3072 bytes)
> [    0.076618] pinctrl core: initialized pinctrl subsystem
> [    0.078003] NET: Registered protocol family 16
> [    0.080995] DMA: preallocated 1024 KiB pool for atomic coherent allocations
> [    0.087419] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
> [    0.087491] hw-breakpoint: maximum watchpoint size is 4 bytes.
> [    0.087635] Serial: AMBA PL011 UART driver
> [    0.090377] bcm2835-mbox 2000b880.mailbox: mailbox enabled
> [    0.129850] bcm2835-dma 20007000.dma: DMA legacy API manager at df80d000, dmachans=0x1
> [    0.132227] SCSI subsystem initialized
[... 221 lines cut ...]
> [ 7325.891398] 8192eu: Unknown symbol wiphy_unregister (err 0)
> [ 7325.899338] 8192eu: Unknown symbol cfg80211_ch_switch_notify (err 0)
> [ 7325.907964] 8192eu: Unknown symbol cfg80211_ready_on_channel (err 0)
> [ 7325.916556] 8192eu: Unknown symbol wiphy_free (err 0)
> [ 7325.923297] 8192eu: Unknown symbol wiphy_new_nm (err 0)
> [ 7325.930502] 8192eu: Unknown symbol cfg80211_get_bss (err 0)
[... cut ...]
> [ 7326.026181] 8192eu: Unknown symbol cfg80211_inform_bss_frame_data (err 0)
> [ 7326.034696] 8192eu: Unknown symbol cfg80211_new_sta (err 0)
> [ 7326.042414] 8192eu: Unknown symbol __cfg80211_alloc_reply_skb (err 0)
> [ 7326.050888] 8192eu: Unknown symbol cfg80211_rx_mgmt (err 0)
> [ 7326.058414] 8192eu: Unknown symbol ieee80211_frequency_to_channel (err 0)
> [ 7326.067251] 8192eu: Unknown symbol wiphy_apply_custom_regulatory (err 0)
> [ 7326.075971] 8192eu: Unknown symbol cfg80211_mgmt_tx_status (err 0)
> [ 7506.202168] 8192eu: Unknown symbol __vfs_read (err 0)

kingosticks, 2018-06-12:
I had a quick look at those instructions and they appear to be for a raspberry pi3. Something compiled for a Pi3 (using kernel7 etc) will not work for a pi1. The other way around does work and that's how everything in Raspbian and Pimusicbox is compiled. There is a single sentence in their document saying to check the raspberry pi website which really isn't very obvious.

2000game, 2018-06-13:
```
Go to the driver’s directory, open the Makefile file to support Raspberry Pi3. By default, the
CONFIG_PLATFORM_I386_PC macro is enabled. Set the value for
CONFIG_PLATFORM_BCM2709 to y, and set the value for
CONFIG_PLATFORM_I386_PC to n. 
```
Can I just use this or do I have to change something? Because in the Makefile of the driver is nothing about `bcmrpi` just `BCM2709`

kingosticks, 2018-06-13:
Have you had a read of https://www.raspberrypi.org/documentation/linux/kernel/building.md ?

2000game, 2018-06-13:
Yes I modified and compiled the kernel like in the tutorial. But in the next step I don't know if I should just use the BCM2709 or change it to bcmrpi.

kingosticks, 2018-06-13:
I think it's pretty clear in my link that Bcm2709 is for a raspberry pi 2 or 3 only. You said originally you were compiling for a raspberry pi 1. You also need to use the raspberry pi specific tool chain, not regular gcc. That's all in the link. I assume the tutorial you are talking about is the tp link document? Perhaps they don't supply instructions or even support bulling for a raspberry pi 1 (since it is more complicated).

kingosticks, 2018-06-13:
I'm afraid I dont have time to dive into this any further right now. If you don't figure it out then leave the issue open and I'll look properly in a couple of weeks.

2000game, 2018-06-14:
Alright thats great, I will wait until then

2000game, 2018-06-14:
It works now, instead searching for the dongle and the pi, I searched for the driver and the pi. I found this [tutorial](https://www.youtube.com/watch?v=cg-ydT13ga8). In the description he had a script that I could use. I just downloaded the [install-wifi](http://downloads.fars-robotics.net/wifi-drivers/) file and used the code from the description. And BAM it worked. And thanks for the tip with the headers.

kingosticks, 2018-06-14:
Oh brilliant. Thanks for the update. I've just this minute ordered this adaptor and I'll get it going myself so we can include the driver in the next release. Did you actually build it or just use the driver at http://downloads.fars-robotics.net/wifi-drivers/8192eu-drivers/8192eu-4.14.26-1099.tar.gz ?

2000game, 2018-06-14:
I used the install-wifi file, it downloaded the driver so yes.