pimusicbox/pimusicbox #484: updatepip.py using deprecaated syntax to get installed packages

Zaat, 2018-06-18:
If you run 'pip install --upgrade pip', the updatepip.py script stops working because of the get_installed_distributions() method, which is no longer there in newer releases.

This is not really the right way to do anyway, I found doing it this way does work.  Here is the updated script (updatepip.py):

`import pip
from subprocess import call
import pkg_resources

for dist in pkg_resources.working_set:
    call("pip install --upgrade " + dist.project_name, shell=True)`

[updatepip.py.txt](https://github.com/pimusicbox/pimusicbox/files/2111007/updatepip.py.txt)

kingosticks, 2018-06-18:
I don't think we use that updatepip.py anymore. I'm happy to delete it entirely

Zaat, 2018-06-18:
If not used, let's remove it then.  Otherwise it does make it a bit confusing for those exploring the internals. :)

kingosticks, 2018-06-18:
I thought maybe you'd found some place I'd forgotten/missed. If you are just exploring then Yeh I'll get rid. Thanks.

commit 2e9f09cf (2018-06-18): Remove unused updatepip file. (Fixes #484)
closed 2018-06-18 by kingosticks