pimusicbox/pimusicbox #486: Spotify Playlists not working since Spotify API Change

thescottyq, 2018-07-11:
Known issue with mopidy-spotify not being able to load the tracks within Spotify Playlists since Spotify updated their API

This will need to be fixed in Mopidy before being ported into pimusicbox, this thread is just to track this issue

paradxum, 2018-07-17:
Can you please verify that this is the issue we are talking about? 
[https://github.com/mopidy/mopidy-spotify/issues/182](https://github.com/mopidy/mopidy-spotify/issues/182)
Does anyone know a work-around to be able to use Spotify at all right now?

kingosticks, 2018-07-17:
> Can you please verify that this is the issue we are talking about?
Yes
> Does anyone know a work-around to be able to use Spotify at all right now?
Use the Mopify web client.

thescottyq, 2018-07-17:
The Mopify Web Client on PiMusicbox does not load Spotify Playlists though. We are waiting on that fix to be rolled into the next build

kingosticks, 2018-07-17:
It doesn't? Why not? Mopify loads playlist data from the Spotify Web API directly. This should work.

rainerda, 2018-07-18:
> Use the Mopify web client.
I'm waaay too stupid to find it :-/

thescottyq, 2018-07-18:
I can confirm that is does not work, even after a rebuild. We have ours set to use Mopify as the default client, we can see the playlist but when we click on it nothing loads

jo3rg, 2018-11-17:
still open?

AlbertMN, 2019-03-19:
Still having this issue,

rainerda, 2019-03-20:
sure, it's a long work in progress:
https://github.com/mopidy/mopidy-spotify/pull/188
tldr; they have to switch to spotify web-api for that
don't bother moving to volumio, same problem there. use spotify-connect in the meantime.

MrGJones, 2019-03-20:
> mopidy/mopidy-spotify#188

I've had same problem with these music apps... found the spotify addon for
kodi works better!

noah1510, 2019-04-08:
I can actually play playlists using spotify connect using a little trick.
First start the playlist on your computer and then select the speaker.
To play something different you first have to select your computer as output again start playing and then select musikbox again.
This workaroud works reliably on v0.7.0RC7