pimusicbox/pimusicbox #491: Logarithmic volume scale for alsamixer

karlvr, 2018-08-11:
I contributed some patches to mopidy-alsamixer to add logarithmic volume scale, but these went in after 1.0.3 and require some GstAudio stuff that doesn't seem to be available in the combination of versions in Pi Musicbox 0.7.

I have made a one-shot patch that patches mopidy-alsamixer to add this capability: https://gist.github.com/karlvr/34e66d2090942dbf9c9bb3c5b2b87260

I've added a comment with a little script (obvious) to install it. I have no idea whether you want to patch a package to add something like this, but if upgrading the package is hard... then perhaps this solves something, if you also care about the volume scale being cubic.

kingosticks, 2018-08-14:
Normally no.. but since I've already done it for [mopidy-spotify](https://github.com/pimusicbox/mopidy-spotify/tree/backport-oauth) (there was no choice) I'm not against also doing it here. Particular since you've kindly gone to the effort of patching it. Was there any particular reason you didn't do the log scale too?

karlvr, 2018-08-15:
@kingosticks thanks mate; I could totally do log too, it's just that I thought that cubic seemed to match what alsamixer did the best!

kingosticks, 2018-08-15:
I won't get a chance to integrate this for a little while so if you are interested in adding log that'd be great. No worries if not, it should be simple enough given what you've already done.