pimusicbox/pimusicbox #492: Web client not accessible from mobile device

mrbinr, 2018-08-29:
hi,
I have a strange Pi MusicBox behaviour, when I use address http://musicbox.local from desktop computer  browser it works, but if i try the same address from my phone it cannot reach to  connect to the page.
Both desktop computer and mobile are on same network.

Best Regards
Pascall

deeweef, 2018-10-31:
Hi Pascall, 
Just had the same problem. In the docs it says that you have to enter the ip-address, but for me just using http://musicbox (without .local) also worked.
Good luck. :)

kingosticks, 2018-11-06:
Android phones do not understand mDNS .local addresses, it's a joke. If your home router is vaguely decent it should resolve http://musicbox (without .local) just fine. If you are not using an Android phone please re-open this.

closed 2018-11-06 by kingosticks