pimusicbox/pimusicbox #494: Cannot use pip

tschuehly, 2018-09-14:
After a fresh install of pimusicbox I ran into issues installing any modules by pip.

```
Collecting python-mpd2
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:318: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#snimissingwarning.
  SNIMissingWarning
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
  Could not fetch URL https://pypi.python.org/simple/python-mpd2/: There was a problem confirming the ssl certificate: [Errno 1] _ssl.c:504: error:14090086:SSL routines:SSL3_GET_SERVER_CERTIFICATE:certificate verify failed - skipping
  Could not find a version that satisfies the requirement python-mpd2 (from versions: )
No matching distribution found for python-mpd2
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
```

The solution should be to update python but I couldn't get it to work.

```
wget https://www.python.org/ftp/python/2.7.15/Python-2.7.15.tgz
--2019-03-16 18:16:16--  https://www.python.org/ftp/python/2.7.15/Python-2.7.15.tgz
Resolving www.python.org (www.python.org)... 2a04:4e42:3::223, 151.101.12.223
Connecting to www.python.org (www.python.org)|2a04:4e42:3::223|:443... connected.
ERROR: The certificate of `www.python.org' is not trusted.
The certificate has expired
```

kingosticks, 2018-09-14:
Hi, is this with the latest version available to download from github? If you really want to try installing the newer version of python you should be able to tell wget to skip the certificate check with a cli option.

tschuehly, 2018-09-14:
Yes this is v0.7.0 Release Candidate 6.

kingosticks, 2018-09-14:
You could try this:
https://stackoverflow.com/questions/49748063/pip-install-fails-for-every-package-could-not-find-a-version-that-satisfies/49748494#49748494

tschuehly, 2018-09-14:
Through some magic it actually worked. 

  ```
  1  wget https://www.python.org/ftp/python/2.7.15/Python-2.7.15.tgz
    2  ls
    3  tar -xzvf Python-2.7.15.tgz
    4  ls
    5  cd ./Python-2.7.15/
    6  ls
    7  ./configure
    8  ls
    9  cat config.log
   10  apt-get install build-essential
   11  sudo apt-get uptade
   12  sudo apt-get update
   13  apt-get install build-essential
   14  ./configure
   15  sudo apt-get install libssl-dev
   16  make
   17  make install
   18  pip install python-mpd2
   19  python -V
```
It returned Python 2.7.15, I also had to  
`pip install --upgrade pip`

Then it worked

kingosticks, 2018-10-12:
Sounds like should be now fixed but I still need confirmation which I will try to do at some point unless someone beats me to it.

kingosticks, 2019-01-08:
https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC7

closed 2019-01-08 by kingosticks