pimusicbox/pimusicbox #497: Set up Spotify Connect and Mopidy

dmckminard, 2018-10-16:
Hey everyone !

I have setup this kind of thing : https://github.com/skalavala/Multi-Room-Audio-Centralized-Audio-for-Home

I was interested in the PiMusicbox but there is no guide about setup an Snapcast server, this is why I've follow the precedent link.

In a post in the Mopidy repository, Mr Kingosticks have speak about how the integration of Mopidy and Spotify connect is made. 
 
"Pimusicbox uses the librespot onevent handler to issue a MPD stop command to Mopidy when librespot playback starts. You could also configure an ALSA dmix device to allow both programs to output to the soundcard at the same time (but you’d still need to remember to stop and start them both to avoid a garbled mess)."

Can someone explain me how to do this kind of thing ?

Thank you all ! Sorry, I'm a novice, and sorry for my English, I'm French !

kingosticks, 2018-10-16:
Did you find an answer?

dmckminard, 2018-10-16:
Yes ! Thank you! Now the only problem is that my music is read very fastly

kingosticks, 2018-10-16:
It's so much more useful that you post the answer that you found rather than just closing the issue. It might help someone in the future. That applies to absolutely any internet forum.

dmckminard, 2018-10-16:
Yes sorry, I’m going to update this!

dmckminard, 2018-10-16:
I've find an answer here : https://github.com/dtcooper/raspotify/issues/30 !

laurensV, 2018-10-24:
> Pimusicbox uses the librespot onevent handler to issue a MPD stop command to Mopidy when librespot playback starts

This is not working for me, it plays whatever is playing through Spotify Connect and Modipy simultaneously

kingosticks, 2018-10-24:
I've already replied to your first (of 3!) posts about your issue.

closed 2018-11-06 by kingosticks