pimusicbox/pimusicbox #499: Connection timeout on web interface unless SSH is connected

icehaunter, 2018-10-19:
Really strange behavior here

Musicbox is definetly up. Can't connect to the web interface, says connection timeout from any device in the network. Try to connect via ssh from one device. It connects, and suddenly web interface is up and running from that device, still ignoring everything else. And I can replicate it from any device in the network.

What's worse is that at some point in the playback it will drop the "connection" again and I need to connect via ssh yet again.

kingosticks, 2018-10-19:
What network connection are you using? Some USB wireless dongles are complete junk with utterly awful "power saving" functionality. 

> What's worse is that at some point in the playback it will drop the "connection" again and I need to connect via ssh yet again.

Can I assume you are talking about losing the web interface connection (rather than the pimusicbox device itself dropping it's connection to the streaming service it's playing).

icehaunter, 2018-10-19:
I am using Ethernet. And no streaming services, only playback from usb

kingosticks, 2018-10-19:
OK. And what Pi is this?

icehaunter, 2018-10-19:
Its a RPi 3 B+ with a 2.5A adapter.
What do you mean with an ip? It only connects with an ip as far as I am aware.

kingosticks, 2018-10-19:
The pi will also have a hostname which by default is set to 'musicbox' and it should also be available at musicbox (and also musicbox.local if your client understands zeroconf/mdns i.e. *not* Android phones).

Couple more questions. Can you try disabling all other optional services, does that have any effect? Can you take a look at the CPU usage? Is this the default web interface?

icehaunter, 2018-10-19:
Describe optional services? I had disabled every and all provider available, didn't touch any settings apart from that and yes, I am using the default web interface

kingosticks, 2018-10-19:
I'm talking about firewall, upnp, shairport, spotify connect, mpd_watchdog. These are all disabled by default so we can forget about those. 

I'm struggling to think what the problem could be here. CPU usage might be interesting when you get a chance.

icehaunter, 2018-10-19:
I don't think CPU has anything to do with it. It accepts ssh connections. It times out on any http connection unless SSH connection has been initiated from the same machine.

Moreover, it exhibits this behavior on a fresh install for me without a single extra file or device present.

kingosticks, 2018-10-19:
The only times I have heard of anything like what you describe was with bad power saving in USB wifi dongles (already ruled out) and CPU issues. I'd rather *know* this has nothing to do with high CPU usage so please help rule that out and we can move on. Thanks.

icehaunter, 2018-10-19:
Ok, of course, I will. I'll have access to that device tomorrow and will keep you posted. Anything else I should collect? Any logs that might point me in the right direction?

kingosticks, 2018-10-19:
I'd have a good hunt around in the normal system log files (/var/log/syslog, dmesg etc) for anything suspicious. You could try also collecting a debug Mopidy log by editing /boot/config/settings.ini. from:
```
[logging]
config_file = /etc/mopidy/logging.conf
```
to:
```
[logging]
config_file = /etc/mopidy/logging-debug.conf
```
Then restart Mopidy (or the whole system) and the Mopidy log will be in /tmp/mopidy-debug.log.

There is a startup log in /var/log/musicbox_startup.log but I don't think that's going to be useful here.

icehaunter, 2018-10-31:
I'm sorry, I had to get the thing working and hadn't the time to debug, so I did a clean archlinux install with mopidy and other stuff I needed and everything works fine. I guess you can close this as non-relevant anymore.

closed 2018-10-31 by kingosticks