pimusicbox/pimusicbox #501: Output whatever plays the Raspberry Line-in

micheledicosmo, 2018-10-23:
I would like to tweak pimusicbox to output whatever input it receives from the line in. I couldn't find this in the manual nor anywhere on the internet, but I think it would be an interesting feature. It could be used both as a mixer between what is being played and another line, or just as 1 line mixer.

I would like to use it for the latter: my pimusicbox output is XLR and this way I could connect an unbalanced stereo to the line in and send the output to my balanced XLR monitors without gaining interference (instead of connecting through RCA). Not sure what bad the line-in quality is tho.

This would also enable users to just connect their phone to the line in via a jack and play the music no matter what. Useful when you need something more reliable than the airplay integration for example.

What do you think?

kingosticks, 2018-10-23:
I assume you mean when using an external soundcard that provides line in, I don't think many (any?) of the audio HATs do.

You would be adding noise and latency when converting your analogue signal back to digital in the external sound device, passing it through the Pi, then converting back to analogue again. What is the benefit of this instead of just using two inputs (one for pimusicbox, one for your phone) on your amplifier and switching between them?

micheledicosmo, 2018-10-23:
@kingosticks Thanks for your answer. My setup would be:

[PHONE/LAPTOP line-out (analog)] -> [PIMUSICBOX line-in (analog)] -> [PIMUSICBOX out (XLR)] -> amplified monitors [XLR in]

Because the monitors are at a distance, I use a balanced connection. I would either need to buy a mixer (the monitors do not need amplification), but would rather save the space (and money) needed for that if that can be accomplished by the pimusicbox.

> I assume you mean when using an external soundcard that provides line in

I wouldn't be using an external soundcard.

kingosticks, 2018-10-23:
What line in input are you planning to use on the raspberry pi?

kingosticks, 2018-10-24:
My point is that the Raspberry Pi does not have an audio line-in socket. So you'll have to use some other external sound device anyway. 

Regardless, I'm afraid this really isn't within the scope of this project.

closed 2018-10-24 by kingosticks