pimusicbox/pimusicbox #504: update to shairport-sync 3.2.2 ?

gibman, 2018-11-06:
The 0.7. Rc6 has an older version of shairport-sync 
Hopefully the next RC will have the latest 3.2.x version ??

I've manually upgraded to shairport-sync 3.2.2 using these steps:

wget https://github.com/mikebrady/shairport-sync/archive/3.2.2.zip
unzip 3.2.2.zip && rm 3.2.2.zip
pushd shairport-sync-3.2.2
autoreconf -i -f
./configure --sysconfdir=/etc --with-alsa --with-avahi --with-ssl=openssl --with-metadata --with-systemv
make && make install
popd
rm -rf shairport-sync*

gibman, 2018-11-06:
I'm using a setup where I have a win10 box with spotify + tuneblade installed.
Tuneblade is able to take the local audio on the PC and re-route this to airplay renderers.

So what I do is; 
On spotify android, I select the spotify connect renderer (located at the win10 box).
Now the spotify windows 10 app starts playing the track.
tuneblade picks up the audio and relays it to the PI's
Now I have in-synch multi room audio in all rooms that have the shairport-sync software running on the pi.

---

The problems are:
Sometimes tuneblade has a connection issue to one of the shairport instances.
Also sometimes the audio disrupts and sounds weird during resyncs.

I would try and let it run for a few weeks on this new version before making any conclusions.
tuneblade could be the culprit as well here.

and 3.2.2 has quite a few relevant and interesting fixes I think :)

kingosticks, 2018-11-06:
Not to mention "Shairport Sync now offers an IPC interface via D-Bus." Nice!

closed 2018-11-06 by kingosticks

kingosticks, 2019-01-08:
Available in https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC7

commit c09eb65b (2018-11-06): Bump shairport-sync to v3.2.2. (Fixes #504)