pimusicbox/pimusicbox #505: sec=ntlm not valid for mount.cifs

djzort, 2018-11-14:
Looks like the option is no longer supported by mount.cifs (version 5.5)

```
root@MusicBox:/opt/musicbox# mount -t cifs -o sec=ntlm,ro,guest //172.29.0.101/tv1/Music /music/Network/
mount error(22): Invalid argument
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
root@MusicBox:/opt/musicbox# mount -t cifs -o ro,guest //172.29.0.101/tv1/Music /music/Network/
root@MusicBox:/opt/musicbox# 

```

editing it out in /opt/musicbox/startup.sh seems to work ok for me

kingosticks, 2018-12-28:
This will be fixed in the next release: https://github.com/pimusicbox/pimusicbox/commit/832a56a997c8e05f2e2e4fc603f4d4828d981a38

djzort, 2018-12-30:
Good fix, thanks.

kingosticks, 2019-01-08:
https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC7

closed 2019-01-08 by kingosticks