pimusicbox/pimusicbox #508: Airplay disappears until reboot after connection or connection failure

conciousweb, 2018-11-17:
After 1 successful or unsuccessful connection airplay will fail to work again until you reboot the device. Any info or fix for this?

Jonic, 2018-12-05:
I think I'm seeing this issue too - would love to know if there's something I can tweak to solve it!

conciousweb, 2018-12-05:
me too, ive found no fix. I have to restart the pi every time after i disconnect

dor666, 2018-12-06:
Try 'mpc status' command in console. Does it time so out?

SplatManDK, 2019-03-05:
I have a similar issue. How can we help the team troubleshoot this?

s4v4g3, 2019-07-27:
I have found that shairport-sync crashes occasionally on connection, but the "monit" service should restart it if it detects that it's not running, so I haven't found that a reboot is necessary to get it working again.  Monit is configured to check at a 2 minute interval, which you can modify by changing the `set daemon xxxx` line in /etc/monit/monitrc.   I changed mine to run at a 10 second interval so the restart would be quicker in case of a crash.   It's also possible that you're running into a different problem where something else gets messed up and can't recover other than by rebooting.

Also, I found that the latest shairport-sync (version 3.3.2) fixes many bugs from the current version running on pi-musicbox (3.2.2) so I have upgraded to the latest and am seeing more stable behavior.

Here are instructions on how to update.  Note that this involves a partial update to debian jessie, which worked fine for me but I haven't tested the entire system afterwards, only the airplay feature.
Modify /etc/apt/sources.list to point to jessie package sources
`# sed -i 's/wheezy/jessie/g' /etc/apt/sources.list`
`# sudo apt-get update`

Install dependencies needed to build and run the latest shairport-sync
`# apt-get install --yes build-essential git xmltoman autoconf automake libdaemon-dev libasound2-dev libpopt-dev libconfig-dev libavahi-client-dev libssl-dev libc6 libconfig9 libdaemon0 libasound2 libpopt0 libavahi-common3 avahi-daemon libavahi-client3 libssl1.0.0 libtool avahi-daemon libsoxr-dev`

Get the source code for shairport-sync at the 3.3.2 tag
`# git clone https://github.com/mikebrady/shairport-sync.git`
`# cd shairport-sync`
`# git checkout 3.3.2`

Configure & build
`# autoreconf -fi`
`# ./configure --sysconfdir=/etc --with-alsa --with-avahi --with-ssl=openssl --with-metadata --with-systemv --with-libdaemon`
`# make`

Kill the shairport-sync process and rename the old version
`# killall shairport-sync`
`# mv /usr/local/bin/shairport-sync /usr/local/bin/shairport-sync-old`

Install the new version
`# make install`

rlp847, 2019-12-14:
Was this only happening on wireless connections or wired as well? Mine constantly did this daily on WIFi, my hard wired ones remained connected. 

The below finally ’fixed’ it, when it cant ping the router it reboots, and it’s set to ping my router every 5 minutes I believe. I’m fairly new to this and this has been driving me crazy, but I now haven’t had to manually reboot one of the 5 Pi’s I have set up for this in about 2 weeks.

 [Link](https://weworkweplay.com/play/rebooting-the-raspberry-pi-when-it-loses-wireless-connection-wifi/)