pimusicbox/pimusicbox #510: Musicbox dissapers from spotify connect after sitting - WARNING  Element doesn't implement handling of this stream. Please file a bug.

jafordjr, 2018-12-28:
Error happens when I let the PI B+ sit overnight and try to access it in the morning. Usually it works but cannot see device on spotify connect afterwards. Sometimes get warning error when I try to access musicbox in browser and do ssh commands, specifically wget.

Often cannot access musicbox browser using ip address using browser after reboot. Have to try to access from different devices before it stops refusing connection...

`    exec('raise tp, value, tb')
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 201, in _actor_loop
    response = self._handle_receive(message)
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 295, in _handle_receive
    return callee(*message['args'], **message['kwargs'])
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 126, in browse
    return self.list_liked()
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 72, in list_liked
    for data in self.backend.remote.get_user_liked():
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/soundcloud.py", line 147, in get_user_liked
    liked = self._get('e1/me/likes?limit=1000')
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/soundcloud.py", line 205, in _get
    res.raise_for_status()
  File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 935, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 500 Server Error: Internal Server Error for url: https://api.soundcloud.com/e1/me/likes?limit=1000&client_id=93e33e327fd8a9b77becd179652272e2
2018-12-27 03:55:15,717 - ERROR    SoundCloudBackend backend caused an exception.
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/mopidy/core/library.py", line 19, in _backend_error_handling
    yield
  File "/usr/local/lib/python2.7/dist-packages/mopidy/core/library.py", line 112, in _browse
    result = backend.library.browse(uri).get()
  File "/usr/local/lib/python2.7/dist-packages/pykka/threading.py", line 52, in get
    compat.reraise(*self._data['exc_info'])
  File "/usr/local/lib/python2.7/dist-packages/pykka/compat.py", line 12, in reraise
    exec('raise tp, value, tb')
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 201, in _actor_loop
    response = self._handle_receive(message)
  File "/usr/local/lib/python2.7/dist-packages/pykka/actor.py", line 295, in _handle_receive
    return callee(*message['args'], **message['kwargs'])
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 126, in browse
    return self.list_liked()
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/library.py", line 72, in list_liked
    for data in self.backend.remote.get_user_liked():
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/soundcloud.py", line 147, in get_user_liked
    liked = self._get('e1/me/likes?limit=1000')
  File "/usr/local/lib/python2.7/dist-packages/mopidy_soundcloud/soundcloud.py", line 205, in _get
    res.raise_for_status()
  File "/usr/local/lib/python2.7/dist-packages/requests/models.py", line 935, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 500 Server Error: Internal Server Error for url: https://api.soundcloud.com/e1/me/likes?limit=1000&client_id=93e33e327fd8a9b77becd179652272e2
2018-12-27 03:55:34,077 - WARNING  404 GET /favicon.ico (192.168.0.108) 13.99ms
2018-12-27 11:28:23,873 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 11:32:35,379 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 11:37:56,384 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 11:42:33,379 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 11:44:46,722 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 11:45:07,852 - WARNING  404 GET /settings/js/mopidy.min.js (192.168.0.108) 14.51ms
2018-12-27 11:45:07,858 - WARNING  404 GET /settings/js/simple.js (192.168.0.108) 3.62ms
2018-12-27 11:45:08,089 - WARNING  404 GET /settings/js/mopidy.min.js (192.168.0.108) 3.69ms
2018-12-27 11:45:08,141 - WARNING  404 GET /settings/js/simple.js (192.168.0.108) 3.67ms
2018-12-27 11:45:08,483 - WARNING  404 GET /favicon.ico (192.168.0.108) 3.95ms
2018-12-27 11:45:45,312 - WARNING  404 GET /favicon.ico (192.168.0.108) 3.74ms
2018-12-27 11:57:59,490 - WARNING  error on read
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 644, in _handle_read
    pos = self._read_to_buffer_loop()
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 614, in _read_to_buffer_loop
    if self._read_to_buffer() == 0:
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 726, in _read_to_buffer
    chunk = self.read_from_fd()
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 1005, in read_from_fd
    chunk = self.socket.recv(self.read_chunk_size)
error: [Errno 113] No route to host
2018-12-27 11:57:59,492 - WARNING  error on read
Traceback (most recent call last):
[... 859 lines cut ...]
2018-12-27 21:24:37,065 - ERROR    Spotify login error: Cannot connect to Spotify
2018-12-27 21:30:40,157 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 21:30:59,611 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 21:32:32,153 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 21:32:35,506 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-27 21:52:47,594 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-28 06:02:27,052 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-28 07:09:33,058 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-28 07:09:40,434 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-28 07:11:32,335 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-28 07:25:02,564 - WARNING  error on read
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 644, in _handle_read
    pos = self._read_to_buffer_loop()
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 614, in _read_to_buffer_loop
    if self._read_to_buffer() == 0:
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 726, in _read_to_buffer
    chunk = self.read_from_fd()
  File "/usr/local/lib/python2.7/dist-packages/tornado/iostream.py", line 1005, in read_from_fd
    chunk = self.socket.recv(self.read_chunk_size)
error: [Errno 113] No route to host
2018-12-28 17:12:19,863 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-28 17:16:33,277 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
2018-12-28 17:20:08,075 - WARNING  Element doesn't implement handling of this stream. Please file a bug.
root@MusicBox:/var/log/mopidy# `

kingosticks, 2018-12-29:
There's so much going wrong here this is difficult to digest. We'd need to
concentrate on one thing at a time (by disabling everything else) to make
progress. But I can say that the specific warning messages are just
warnings and are not causing issues here.

Spotify connect functionality is implemented in librespot, it's entirely
separate to Mopidy. Librespot should have its own log file which may help
track down the issue. When librespot crashes, as it often does, it should
restart automatically. At which point you may need to properly restart your
Spotify client (running on your phone etc) for it to work out what's
happened.

It also looks like there are some general network connectivity issues going
on here and that might be what's affecting librespot. Or some of the other
issues in the Mopidy log. Is this a wired or wireless connection? Do you
have a reliable power supply?

On Fri, 28 Dec 2018, 23:31 jafordjr <notifications@github.com wrote:

> Error happens when I let the PI B+ sit overnight and try to access it in
> the morning. Usually it works but cannot see device on spotify connect
> afterwards. Sometimes get warning error when I try to [...]

smeusekentik, 2019-01-28:
Had the same thing happen to me last night. I noticed a "connection reset by peer" message in the log, but it seems like Librespot was able to re-authenticate:

`WARN:librespot_playback::player: Track "The Ecstasy of Gold" is not available
INFO:librespot_playback::player: Loading track "Blackened (Live - Seattle '89)" with Spotify URI "spotify:track:2PFoYVONNUOm7HHkTAJers"
INFO:librespot_playback::player: Track "Blackened (Live - Seattle '89)" loaded
INFO:librespot::player_event_handler: Running ["/opt/musicbox/mpc_stop.sh"] with environment variables {"TRACK_ID": "5d185330a9004fd38c9f04edf2a0ea56", "PLAYER_EVENT": "change", "OLD_TRACK_ID": "f26113ac557f463f8690671b3021cd42"}
INFO:librespot::player_event_handler: Running ["/opt/musicbox/mpc_stop.sh"] with environment variables {"PLAYER_EVENT": "stop", "TRACK_ID": "5d185330a9004fd38c9f04edf2a0ea56"}
ERROR:librespot_core::session: Error { repr: Os { code: 104, message: "Connection reset by peer" } }
INFO:librespot_core::session: Connecting to AP "guc3-accesspoint-b-69l7.ap.spotify.com:4070"
INFO:librespot_core::session: Authenticated as "redacted" !
INFO:librespot_playback::audio_backend::alsa: Using alsa sink
INFO:librespot_core::session: Country: "US"`

It shows up in the client grayed out with the message "can't play this right now". Restarting the client doesn't help. I'll avoid rebooting Musicbox to see if there is some other info I can find.

kingosticks, 2019-01-28:
Is it all tracks? Or just this one? There's an on-going librespot issue where (seemingly) depending on your region and the particular track song, Spotify wants to serve the audio from it's CDN - which librespot does not support. Can you try a few tracks?

smeusekentik, 2019-01-28:
Interesting. It seems like it's only Podcasts at the moment. Music seems to be working. Will try to do some packet captures to see what else I can find.

kingosticks, 2019-01-28:
Ahh, librespot does not support podcasts yet. They work differently to everything else.

smeusekentik, 2019-01-28:
Heh, ok, nevermind :)

smeusekentik, 2019-02-10:
Made one change, seems to be behaving for the moment. I changed my login info from the email address to the actual username (long string). It lasted 24hrs, which is the longest yet. Will keep an eye on it.