pimusicbox/pimusicbox #513: Musicbox doesn't boot on Raspberry Pi 3 A+

DJC27, 2019-01-07:
I have tried to install Pi Musicbox on my new Rpi 3 A+. However it doesn't even start the booting process, but gets stuck on the rainbow image. The Pi and the SD Card definetly work (tested with the latest rasbian lite and a Rpi 2) and the Musicbox image works on the on the Rpi 2. However when I insert the SD Card into the 3 A+ model, it gets stuck. So I'm not really sure where to go from here... Or is the Rpi 3 A+ simply not supported?

kingosticks, 2019-01-07:
I'm afraid the latest image predates the release of the model 3 A+ and the required firmware. I did an update over Christmas but then somehow failed to release it. I will do that. 

Alternatively you can boot pimusicbox using your model 2 and update the firmware using that. Then  swap the SD card back to the model 3 A.

kingosticks, 2019-01-08:
There's https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC7 available now with a sufficiently updated kernel. Since I don't have a 3 A+ I would be very grateful if someone could test it and report back.

closed 2019-01-08 by kingosticks

DJC27, 2019-01-08:
I have tried it this morning. It works like a charm. Thank you very much for your quick response! That was some serious "customer service", well done! :)

kingosticks, 2019-01-08:
Glad to hear that, thanks a lot for testing it. I should have released it ages ago so I really can't accept any praise!

commit a19530bf (2019-01-08): Bump kernel to 4.14.89. (Fixes #513)

Updated brcm wireless firmware.