pimusicbox/pimusicbox #516: Creating an image

grawlybean, 2019-01-23:
I am trying to avoid creating a new SD card for RC 7 so trying to follow "Creating an image" but Step 5 does not work indicating initially
losetup: invalid option --p
followed by various mount errors etc.

Any help
Thanks
Paul

kingosticks, 2019-01-23:
The version of `losetup` available in [Debian](https://manpages.debian.org/stretch/mount/losetup.8.en.html) and [Ubuntu](http://manpages.ubuntu.com/manpages/bionic/man8/losetup.8.html) have this option. I guess you have a different version. Is there something wrong with the released RC7 [available to download](https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC7) ?

grawlybean, 2019-01-23:
Thanks for the reply.
No I am sure the RC7 release is great. I was just trying to update my existing v0.6 rather than install to a new SD card.

I have used Pi Musicbox for a while now and also Raspotify but the Spotify Connect is now failing hence why I am updating to the latest version.

Thanks for all your efforts with this great software.

kingosticks, 2019-01-23:
There is no update to Spotify Connect in RC7 ([changelog](https://github.com/pimusicbox/pimusicbox/commit/2896a43c9f9dbb64cfec6b5d03d92bc4eebac20f#diff-fed77448f29b992f3ad4fc8670ec3812)), I don't think this will help you. The Spotify Connect software is buggy.

grawlybean, 2019-01-23:
Yes I thought that would be the case so was planning to revert to using the Pi Musicbox webinterface instead much as I love Spotify Connect... when it works. Thanks again.

kingosticks, 2019-01-23:
Sadly the playlists are still broken in the Pi Musicbox interface so you are really limited to searching (or creating your own non-spotify playlists). You could switch to use the Mopify interface, I *think* that has working playlist support.

grawlybean, 2019-01-23:
ok thanks

closed 2019-01-23 by grawlybean