pimusicbox/pimusicbox #517: Not playing .WAV files

Paul-vk3tgx, 2019-01-31:
With Pi MusicBox 0.7.0RC7 ver, it lists my .wav files when I browse my music on a NAS, however when I try and play them MusicBox's 'Now Playing' correctly lists the track, but the progress bar says '(sync)' rather than a bar, and there is no audio.
.mp3 & .flac files on the save NAS play ok.

kingosticks, 2019-01-31:
Can you please provide the Mopidy log files. Details at http://docs.pimusicbox.com/en/latest/troubleshooting/

Paul-vk3tgx, 2019-02-01:
As a further, I tried rebuilding it on a fresh SD card - no change.
I also copied a sample wav file to the /music/MusicBox folder - same fault

And for the log files
[mopidy.log](https://github.com/pimusicbox/pimusicbox/files/2821557/mopidy.log)
[musicbox_startup.log](https://github.com/pimusicbox/pimusicbox/files/2821561/musicbox_startup.log)
[mopidy-debug.log](https://github.com/pimusicbox/pimusicbox/files/2821564/mopidy-debug.log)

Paul-vk3tgx, 2019-02-01:
Nick,

Thanks for replying – I’ve just uploaded the .log’s to GitHub.

Paul.

From: Nick Steel
Sent: Thursday, January 31, 2019 8:59 PM
To: pimusicbox/pimusicbox
Cc: Paul-vk3tgx ; Author
Subject: Re: [pimusicbox/pimusicbox] Not playing .WAV files (#517)

Can you please provide the Mopidy log files. Details at http://docs.pimusicbox.com/en/latest/troubleshooting/