pimusicbox/pimusicbox #521: Creating an image

Flip1024, 2019-03-04:
Hello,
I'm trying to create a new image, but I get following error:

pi@raspberrypi:~/mbsrc $ sudo ./src/makeimage.sh musicbox.img bigger
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git
./src/makeimage.sh: Zeile 20: bc: Kommando nicht gefunden.
./src/makeimage.sh: Zeile 26: [: -lt: Einstelliger (un?rer) Operator erwartet.
INFO: Enlarging musicbox.img...
truncate: ung?ltige Zahl: ?musicbox.img?

Welcome to fdisk (util-linux 2.29.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): Partition number (1,2, default 2): 
Partition 2 has been deleted.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (2-4, default 2): First sector (2048-1976319, default 2048): Last sector, +sectors or +size{K,M,G,T,P} (122880-1976319, default 1976319): 
Created a new partition 2 of type 'Linux' and of size 905 MiB.
Partition #2 contains a ext4 signature.

Command (m for help): 
The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.

e2fsck 1.43.4 (31-Jan-2017)
Durchgang 1: Inodes, Bl?cke und Gr??en werden gepr?ft
Durchgang 2: Verzeichnisstruktur wird gepr?ft
Durchgang 3: Verzeichnisverkn?pfungen werden gepr?ft
Durchgang 4: Referenzz?hler werden ?berpr?ft
Durchgang 5: Zusammengefasste Gruppeninformation wird gepr?ft
/dev/loop0p2: 31273/65536 Dateien (2.1% nicht zusammenh?ngend), 189796/231680 Bl?cke
resize2fs 1.43.4 (31-Jan-2017)
Das Dateisystem ist bereits 231680 (4k) Bl?cke lang. Nichts zu tun!

INFO: Increased musicbox.img from 965M to 965M

pi@raspberrypi:~/mbsrc $ ./src/chroot.sh musicbox.img create_musicbox0.7.sh
Info: Checking have root access to mount the disk images.
Mounting musicbox.img and preparing arm chroot...
exit 101
Executing 'chroot rootfs /bin/bash -c ./tmp/create_musicbox0.7.sh'
************************************************
** ERROR: Insufficient free space to upgrade  **
** Use ./makeimage.sh bigger <image_file>     **
************************************************
Cleaning up chroot and unmounting...
Unmounting /home/pi/mbsrc/rootfs/tmp
Unmounting /home/pi/mbsrc/rootfs/sys
Unmounting /home/pi/mbsrc/rootfs/proc
Unmounting /home/pi/mbsrc/rootfs/dev/pts
Unmounting /home/pi/mbsrc/rootfs/dev
Unmounting /home/pi/mbsrc/rootfs/boot
Unmounting /home/pi/mbsrc/rootfs
pi@raspberrypi:~/mbsrc $ 

There is insufficient free space. Is there a script error?

kingosticks, 2019-03-04:
Releases can be downloaded from the releases tab on github. That script is not designed to l be run on raspbian (or a raspberry pi at all).

Flip1024, 2019-03-04:
Thanks for your reply.
How can I create a release für my raspberry pi?

Regards

Flip1024, 2019-03-04:
my problem is, that I will run the pi as access point, in order to use it as mobile MP3 player and manage it over my cellphone

maybe you have a tutorial to this - all I tried, the never worked, except with a fresh rasping image

Flip1024, 2019-03-12:
Hello
I've solved the problem, with updating to Debian "squeeze".
 
First I took the preconfigured image.
After that I updated to "squeeze".
Then I configured the pi as access point.
Everything worked.

Regards Christian

closed 2019-03-12 by Flip1024