pimusicbox/pimusicbox #522: 500: Internal Server Error

SplatManDK, 2019-03-05:
On vanilla install (only changes WiFi settings), selecting the "Settings" section in the browser UI returns: 500: Internal Server Error

SplatManDK, 2019-03-05:
It seems the problem could related to editing the config file in Windows. Probably a "bad" thing to do, but I can't possibly be the only n00b user who did this. So if this is indeed the problem then perhaps a warning in the instructions would be prudent, along with instructions on how to edit the file "correctly".

simba83, 2019-03-06:
There is no problems editing this file under Windows, but it depends on how you edit and save the file. Depending on your application you choose to use there sertainly can be errors.
Use a good editor like notepad+ and there should be no problems.

SplatManDK, 2019-03-09:
I did use Notepad++ so that's kind of odd. I re-flashed the SD card just to be sure, and even though Etcher valided the first card correct, it's now solved. So clearly a corruption issue. Let's close this ticket, and I'll re-open a new one if it occurs again, and I do some more digging into the actual cause. :-)

closed 2019-03-09 by SplatManDK