pimusicbox/pimusicbox #523: Playing songs with spotify connect

Ivaneres, 2019-03-28:
It is currently not possible to start playing a different song when another is already playing when using Spotify connect; instead it must first be added to the song queue and the currently playing song skipped.

My apologies if this has already been pointed out.

kingosticks, 2019-03-28:
This is a bug in librespot that's since been fixed but we are still shipping the old, bugger verison. I need to update librespot. Thanks for the reminder.

fiid, 2019-04-05:
I fixed this on my MusicBox like this...

```
ssh root@musicbox.local
[default pw is musicbox]

curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo apt-key add -v -
echo 'deb https://dtcooper.github.io/raspotify jessie main' | sudo tee /etc/apt/sources.list.d/raspotify.list
apt-get update
apt-get -y install raspotify
rm /opt/librespot/librespot
ln -s /usr/bin/librespot /opt/librespot/librespot
reboot
```

mattbot101, 2019-05-19:
My workaround for this is to click the connect to device button on PC, choose this computer, start playing the song you want, then switch back to the device. ("listening on MusicBox") However when I get some time I'll probably use @fiid 's solution, thanks!

Clayzway, 2019-05-25:
> 
> 
> I fixed this on my MusicBox like this... commands in terminal...
> 
> ```
> ssh root@musicbox.local
> [default pw is musicbox]
> 
> curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo apt-key add -v -
> echo 'deb https://dtcooper.github.io/raspotify jessie main' | sudo tee /etc/apt/sources.list.d/raspotify.list
> apt-get update
> apt-get -y install raspotify
> rm /opt/librespot/librespot
> ln -s /usr/bin/librespot /opt/librespot/librespot
> reboot
> ```

hero!

pakocg, 2019-05-31:
> I fixed this on my MusicBox like this... commands in terminal...
>
> ```
> ssh root@musicbox.local
> [default pw is musicbox]
>
> curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo apt-key add -v - [...]

Thank you very much, it worked for me also :))

fiid, 2019-06-28:
Glad it's working!!  👍 

Note you can re-run the two apt-get commands to upgrade raspotify if anything gets funky in the future.

sweets9, 2020-04-28:
I am having trouble with spotify connect where the player appears in list but spotify cannot connect. Is this fix still working? I noticed that some sources are 404'ing now although raspotify installs OK.

Anyone able to help? Thanks!

HBoardman, 2021-03-06:
> I fixed this on my MusicBox like this... commands in terminal...
>
> ```
> ssh root@musicbox.local
> [default pw is musicbox]
>
> curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo apt-key add -v - [...]

Worked for me, absolute saviour.