pimusicbox/pimusicbox #525: Force HTTPS conection

favy87, 2019-04-05:
Forzar la conexion HTTP a HTTPS.

Force HTTP to HTTPS

kingosticks, 2019-04-05:
Pimusicbox is designed to be used within your trusted network, that should have been made clear in the documentation. I'm going to need more than 4 words here else this will be closed as to me this seems outside the scope of the project.

favy87, 2019-04-05:
Hola, le hemos pasado un test de vulnerabilidades a la versión 6 (openvas) y salen varias vulnerabilidades. Una de ellas es que debería de obligar a qué la conexión sea https ya que si no las contraseñas viajan en texto plano.

kingosticks, 2019-04-05:
I don't speak Spanish so I'm relying on a translation tool, perhaps I misunderstand your reply because you haven't addressed my point. I'm also at a loss as to what certificate we could use to achieve what you want. If you have some more thoughts (in English) please feel free to reopen this.

closed 2019-04-05 by kingosticks