pimusicbox/pimusicbox #527: Buttons in Browser Incorrect when Casting to MusicBox from WMP

jackho9, 2019-05-21:
Love MusicBox.  I noticed two issues with it when casting to MusicBox from Windows Media Player (UpMPD):

1) The skip song back/forward buttons in the browser cause the warning shown in the attachment ('Problem looking up https:/...') and don't work; while the pause button in fact skips to the next song.

2) When it switches from one song to the next in the list it generates the Warning with 'Please file a bug', but it still works fine.

Cheers!
Jack
[image]

kingosticks, 2019-05-21:
Thanks, it's nice to hear when the project has a (happy) user!

Just so I understand this right, when you use the "Next" and "Previous" buttons in the default webclient, they don't do anything? And if you press the "Pause/Play" button it skips. I think this might be related to the poor handling of wma files in the old version of GStreamer that PiMusicbox uses. Are you able to try some mp3 files?

jackho9, 2019-05-22:
Hi Nick, correct.  I was streaming FLAC files, if that makes any
difference.  I will try mp3s when I have a moment.

Jack

jackho9, 2019-05-22:
Ok, I am having difficulty confirming the exact nature of the issue.  This is what I can confirm:

1) When I 'cast to MusicBox' from Windows Media Player it plays mp3s and flacs but not wav/wma (in this last case it says that GStreamer is missing a plugin - that's fine I don't play wavs, I just tried a few now for completeness).  When casting to MB, the next/previous and pause buttons in the default webclient have unexpected behavior that I am having difficulty pinning down, because it seems to change  (it's as if they do not speak the same language)

2) When I play things directly from the webclient (radio, mp3/flac files or from Google Music) the buttons behave as expected.

3) Almost every time I change a track I get the Warning with 'Please file a bug' but it plays it fine.

4) The start of the session looks as follows (says the version of Python 2.7 installed is deprecated):

[image]

kingosticks, 2019-05-22:
OK so the issue is that the webclient controls have the incorrect behaviour when you are playing music from a Windows PC via upmpdcli. In the websettings there is an option to enable debug logging. Can you do that and provide the full debug log that shows the buttons doing the wrong thing.

jackho9, 2019-05-24:
Ok, here is the debug log.  I turned on the Zero, selected some flac files in WMP and cast them to MusicBox.  When they started playing  I played with the next/pause/previous buttons in the 'Playing Now' page of the webclient, bith on my phone and computer.  Sometimes they work, sometimes they don't, pausing skips, then it hangs.  At that point I killed WMP and started the same album from Google Music, all good there, still playing.

Jack

[mopidy-debug.log](https://github.com/pimusicbox/pimusicbox/files/3217533/mopidy-debug.log)