pimusicbox/pimusicbox #529: PiMusicBox Raspberry Pi 4 support?

schadha, 2019-07-05:
Is this supported for the newest Raspberry Pi 4 model? I haven't been able to boot into it using the new model.

kingosticks, 2019-07-05:
Nope, afraid not. It needs the new /boot software.

schadha, 2019-07-06:
@kingosticks Any plans to add that support in the near future?

kingosticks, 2019-07-06:
I will try next week but I'm doubtful it'll be as simple as running the existing scripts on the new image. Let's see what happens.

robertgarrigos, 2019-12-26:
Any news on this?

mattkc93, 2020-04-21:
Any news on rpi 4 support? Just bored during quarantine

KuDeTa, 2020-07-05:
Another vote for this. Happy to help if we could get some direction (never used the project before). @kingosticks

tobiasb, 2020-08-20:
Same here, any news?

churaux, 2020-09-22:
Another vote for this enhancement. it's complicated to adapt to raspberry 4 ? 
Many thanks

boingo-00, 2021-01-28:
Distro update probably should solve firmware problem

nikosd, 2021-02-06:
My vote on this as well!

👍

Arakade, 2021-04-22:
+1 (sorry if this isn't the right way to vote :) )

kingosticks, 2021-04-22:
I have no plans to do this. I am aware of how outdated the software is and how the lack of Pi 4 support is disappointing but I'm just not interested any longer. This project needs a new maintainer.

Arakade, 2021-04-22:
> I have no plans to do this. I am aware of how outdated the software is and how the lack of Pi 4 support is disappointing but I'm just not interested any longer. This project needs a new maintainer.

Understood.  I've only just discovered the project so I have no idea on the following but thought it might be worth asking:

Roughly what would be involved?

My assumption is that this project might be built somewhat akin to a Docker file -- layering stuff atop the Raspian image, right?  So I guess it's a question of needing to update the base image then...?  If not the case, perhaps some pointers to get _someone_ started?

Thanks for thoughts (and previous work!)

kingosticks, 2021-04-23:
That's essentially it. You can see https://github.com/pimusicbox/pimusicbox/blob/develop/create_musicbox0.7.sh

But it was based on a really, really old Raspbian base image. So much has changed since then. It would be an exercise in looking through what's there, translating it to the new way Raspbian does it and also looking if the change even still makes sense. Or starting over from scratch.