pimusicbox/pimusicbox #530: FTP Server on pimusicbox

TrubshawGit, 2019-07-11:
Have been trying to install an ftp server on my piMusicBox with no luck.
I run apt-get update and then apt-get install proftpd-basic but nothing.
I've tried this ssh and by attaching a screen and keyboard to the Rpi, same result each time.
Attached is the terminal output, you will see that update has errors.
Any suggestions?
Thanks,
TGit
[proftpd.txt](https://github.com/pimusicbox/pimusicbox/files/3381491/proftpd.txt)

kingosticks, 2019-07-11:
> Err http://mirrordirector.raspbian.org wheezy/main Sources
  404  Not Found [IP: 93.93.128.193 80]

I had a very quick poke around but it seems they have finally removed the Raspbian Wheezy package repositories. Which is fair enough, it has been years. But unfortunately that means you cannot update or install any software through APT.  This is a terrible state of affairs which I only have my own laziness to blame. Sorry.

If you are desperate, as a quick workaround you can use pyftpdlib:
```
pip install pyftpdlib
python -m pyftpdlib -i :: -d /some/directory -u bob -P secret
```
That'll run an FTP server on port `2121` with username `bob` and password `secret`.

Looks like I know what I'm doing this weekend.

TrubshawGit, 2019-07-11:
OK
I've used python -m pyftpdlib
I will have to ssh in to run it but at least I can get there now headless.

TrubshawGit, 2019-07-11:
Thanks very much kingo. Don't waste time on this on my behalf.

kingosticks, 2019-07-11:
It's awful I have let the software become so outdated they stopped providing the packages. Thanks for bringing this to my attention, if this isn't the push I need I don't know what would be!

s4v4g3, 2019-07-27:
I was able to proceed by just changing "wheezy" to "jessie" in sources.list:
`# sed -i 's/wheezy/jessie/g' /etc/apt/sources.list`
`# apt-get update`

I didn't try performing a full upgrade after this, but was able to install all the packages I needed, like git, build-essential, etc. & didn't run into any problems.  During the installation of one of the packages I was prompted for what to do with one of the configuration files (avahi I think) and I accepted the default option to leave the file as is.

@kingosticks I don't mean to take away any of your motivation to update to Buster.  I am happy to help out if you need any work done.  This is a really useful project!

TrubshawGit, 2019-07-27:
> I was able to proceed by just changing "wheezy" to "jessie" in sources.list:
> `# sed -i 's/wheezy/jessie/g' /etc/apt/sources.list`
> `# apt-get update`

Thank you. I've made the change and installed an ftp server. That's all I wanted, so thank very much.
TGit

kingosticks, 2019-07-27:
I think you might find thay breaks some components of pimusicbox but if it
works good enough for you then great. Thanks for the tip.

IvanMCheng, 2019-10-30:
At safer change might be to use:
deb http://legacy.raspbian.org/raspbian/ wheezy main contrib non-free rpi
deb-src http://legacy.raspbian.org/raspbian/ wheezy main contrib non-free rpi
This will make it possible to add packages from the wheezy repo to the base os. 
The www.lesbonscomptes.com repository will still fail...

revolunet, 2019-11-11:
Hi, what is the www.lesbonscomptes.com dependency ?

revolunet, 2019-11-11:
Ok looks like is for the `upmpdcli` package

AurelFR, 2021-01-04:
type on ssh console 
sudo nano /etc/apt/sources.list
delete everithing there 
then 
deb http://deb.debian.org/debian/ jessie main contrib non-free
deb-src http://deb.debian.org/debian/ jessie main contrib non-free

deb http://security.debian.org/ jessie/updates main contrib non-free
deb-src http://security.debian.org/ jessie/updates main contrib non-free

deb http://deb.debian.org/debian/ jessie-updates main contrib non-free
deb-src http://deb.debian.org/debian/ jessie-updates main contrib non-free
then 
sudo apt-get update 

working for me /i just install moc player on whezzy 

have fun /pimusicbox rocks on 2021