pimusicbox/pimusicbox #534: Librespot without a fixed user?

Saik19922, 2019-08-20:
I've setup my musicbox and so far it's mostly working good. 

However I've set it up for both me and my girlfriend living with me. We are both using it, with different Spotify Accounts. She has a Facebook-created account so logging in with user credentials is very tedious if not impossible - so I've used my username and password for the web-search functions and spotify features of Pi MusicBox.

However, we both occassionally use it via Spotify Connect, and she can't since a restart always changes the /etc/default/librespot file's ARGs to use my username and password.

How can I change it to accept any user?

kingosticks, 2019-08-20:
Librespot does support non-hardcoded credentials, I think. But I've never tried it.

To use it you would remove the username and password arguments that are present in /etc/default/librespot. Once librespot is restarted it should login automatically when each user selects it on their Spotify client. I don't know if it works for Facebook logins.

If it does work reliably, we could make that the default. Or, you can hack your version to do it.

Saik19922, 2019-08-20:
I am removing the username and password arguments in /etc/default/librespot but upon a restart Pi MusicBox will repopulate the file with my username and password.

It does work without username and password, I've been able to confirm that. But as of now I have to edit /etc/default/librespot every restart.

kingosticks, 2019-08-20:
OK, then comment out the line in /opt/musicbox/startup.sh that writes that librespot settings file.

bluepuma77, 2020-03-19:
I think Facebook-created Spotify accounts also have a regular username. Just log in and check.

nikaiser, 2021-04-03:
Fo future users, I can confirm this works:

Step 1. Go to Spotify: https://accounts.spotify.com/en/login/
Step 2: Use Facebook account to log-in
Step 3: Go to Account Overview
Step 4: Under "Profile", your _Username_ should show up as a numeric string.
Step 5: Use this _Username_ in the /config/settings.ini file and your usual password.

**Tip:** I had to reset my password to get this to work as I don't think I had a non-facebook login password set.

Hope this helps.