pimusicbox/pimusicbox #535: SSL error on spotify/token

ivanyv, 2019-11-24:
Hi, I'm getting this error on 0.7.0RC7 when searching on Spotify:

```
ERROR    Fetching https://auth.mopidy.com/spotify/token failed: HTTPSConnectionPool(host='auth.mopidy.com', port=443): Max retries exceeded with url: /spotify/token (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",),))
```

kingosticks, 2019-11-24:
Can you check the system date and time are correct. I think there's a thread about this issue on http://discourse.mopidy.com

ivanyv, 2019-11-24:
Ah, that's it. Would be great if date was set via NTP.

closed 2019-11-24 by ivanyv

kingosticks, 2019-11-24:
It is supposed to be , but for some reason it doesn't seem to always work.
Perhaps it's happening too early in the boot process and the network is
guaranteed to be ready yet. It's something that needs looking into for
sure.