pimusicbox/pimusicbox #537: Can't install Mopidy-Touchscreen

TheBurnoutTV, 2020-02-16:
I'm trying to install Mopidy Touchscreen but everytime I try it gives me this error: 

```
/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/primitives/constant_time.py:26: CryptographyDeprecationWarning: Support for your Python version is deprecated. The next version of cryptography will remove support. Please upgrade to a 2.7.x release that supports hmac.compare_digest as soon as possible.
  utils.DeprecatedIn23,
/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
  utils.CryptographyDeprecationWarning
Collecting Mopidy-Touchscreen
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",),)': /packages/f4/ab/5311965d4b9609322e2baa42ad7d2fc63fa80e8c1199f41f953787920455/Mopidy-Touchscreen-1.0.0.tar.gz
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",),)': /packages/f4/ab/5311965d4b9609322e2baa42ad7d2fc63fa80e8c1199f41f953787920455/Mopidy-Touchscreen-1.0.0.tar.gz
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",),)': /packages/f4/ab/5311965d4b9609322e2baa42ad7d2fc63fa80e8c1199f41f953787920455/Mopidy-Touchscreen-1.0.0.tar.gz
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",),)': /packages/f4/ab/5311965d4b9609322e2baa42ad7d2fc63fa80e8c1199f41f953787920455/Mopidy-Touchscreen-1.0.0.tar.gz
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",),)': /packages/f4/ab/5311965d4b9609322e2baa42ad7d2fc63fa80e8c1199f41f953787920455/Mopidy-Touchscreen-1.0.0.tar.gz
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/f4/ab/5311965d4b9609322e2baa42ad7d2fc63fa80e8c1199f41f953787920455/Mopidy-Touchscreen-1.0.0.tar.gz (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'SSL3_GET_SERVER_CERTIFICATE', 'certificate verify failed')],)",),))

You are using pip version 18.1, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

When I try to upgrade pip it gives me the same error. I'm new to this kind of stuff so I hope anyone has an idea how to fix this.

rhmk, 2020-04-06:
Hi,
as the error sais "certificate verify failed", so you can try running pip  with "--trusted-host files.pythonhosted.org". 
See https://pip.pypa.io/en/stable/reference/pip/#general-options:

```
--trusted-host <hostname>¶
Mark this host or host:port pair as trusted, even though it does not have valid or any HTTPS.
```

TheBurnoutTV, 2020-04-07:
> Hi,
> as the error sais "certificate verify failed", so you can try running pip with "--trusted-host files.pythonhosted.org".
> See https://pip.pypa.io/en/stable/reference/pip/#general-options:
>
> ``` [...]

Thanks, that works but now I have another problem. After trying to install Mopidy Touchscreen it gives another error with the final sentence: 
`Unable to run "sdl-config". Please make sure a development version of SDL is installed.
`
I found https://pypi.org/project/PySDL2/ and I tried to download a dev version of SDL but I can't download it with git. I tried to install that as well but it just says that it is unable to locate package.