pimusicbox/pimusicbox #539: RPI 1 model B + pimusicbox 0.7 + hifiberry old DAC

7091lapS, 2020-03-27:
Hi, I'm trying to repurpose an old model B to radio player and I have an old version of the very first hifiberry DAC that was intended for model B. I'm using pimusicbox and although I've configured the output settings to use Hifiberry DAC/DAC+/etc.. I can't get the any sound. Everything works fine if I play through the normal headphone jack, so I suspect it might be a pimusicbox compatibility issue with my old DAC. Can anyone help with this? Or know what is the last pimusicbox version to support the original hifiberry DAC?
Cheers

kingosticks, 2020-03-27:
I think I have the same dac kame where. I didn't intend to remove support for that dac at any point. Could you post your startup log

http://docs.pimusicbox.com/en/latest/troubleshooting/

kingosticks, 2020-03-27:
Actually, since you already cross posted let's answer this on the forum.

closed 2020-03-27 by kingosticks