pimusicbox/pimusicbox #543: Change image creation towards  CustomPiOS

MaStr, 2020-05-23:
Hey,
recently, I stumbled again over your nice pimusicbox.
Unfortunately, I can't build the system like I want it to be because of the very old Raspbian. So my friend will receive "only" a vanilla pimusicbox.

Same like the last time I used this project, the build path of the image is using an already existing image to create v0.7. So, it is not totally clear what is needed and what not.

For PirateBox, I used my own set of script to build ArchLinux images, but that would be overkill to port an debian based distribution to ArchLinux.

But, I am using octoprint on another RPi, too. This is using [octopi](https://github.com/guysoft/OctoPi) , which is a Raspbian based distribution. The image is created with [CustomPiOS](https://github.com/guysoft/CustomPiOS). This is a framework for repeatable builds. 

I am wondering if it makes sense to put some work in there together with you guys, to change the build to a more modern Debian image? (Asking, because it won't make sense to work alone in a box for days/hours and then nobody wants to adapt this).

Best regards Matthias

kingosticks, 2020-05-23:
Hi Matthias,

The release is horribly out of date, as you've been polite to say. I've pretty much stopped maintaining it as I stopped using it so it's not in a good way these days. Arguably I should have made this clearer but it still seems to be useful for some people so I've left things how they are. I would like to remedy the poor state and I'd be happy to almost start-over rather than try and work out what we still need - we are of course two releases behind now and many things have moved on considerably. I'd welcome anyone who has some experience and was interested in helping with an update. I spent some time looking at CustomPiOS before and I thought it looked quite complicated but I will take another look. I also didnt want to restrict things to Raspbian so I've also thought the updated version would be based on regular Deb packages. Some people don't want the whole system so a scheme that allowed people to pick and choose would be nice.

MaStr, 2020-05-23:
Hi, 
thank you for your replying so quickly. I noticed the state of the project based on the issues ;-)
In fact, the old distribution is one of the reasons for like 50% of the newer issues :-1: 

My major concern in porting the current repository into <something> is, that there are a collection of etc configuration-content.. and it is hard to understand on a first shot what is intended to be changed or what is only default configuration.

I understand you would like to move forward toward a "simple" .deb package. Before you can go that route, you need to get all the stuff sorted. 

Beside of the configuration, the other things are the distribution changes... like "do we still monit" or is systemd enough ..  *cough* *cough* *cough* (I like monit).

I just started to throw the tings together for CustomPiOS, it takes some time to understand.. but using the octopi repository as a template/help/cheat sheet .. seems to help (nothing build yet, tbh).

> Some people don't want the whole system so a scheme that allowed people to pick and choose would be nice.
Oh, I feel that. TBH: I ignore it. If you don't you end up like PirateBox: having all custom configuration in /opt/myproject . In your case, you can commit yourself and stay at the point of "being a special purpose image". 
In fact, I can feel totally the pain of not using my <other board brand> board, like OrangePi. As far as I understand the customPiOS, you have a schema which you can use on a clean ubuntu or armbian, too. So, that should help. Another point is, that after "documenting" the installation process in a new build pipeline, you have a first step to a general installation instruction.

best regards Matthias

satzg, 2020-05-23:
I still use the original version and it works fine for what it does. Did you stop streaming music or did you move to something different?

Thanks,
Greg

kingosticks, 2020-05-23:
I moved and changed my media setup. I do still have a use for a system like pimusicbox but I've spent the time elsewhere.

I think I'd be one of the people that would value the pick-and-mix option but it does really multiply out the support effort and that's something I was, at times, spending far too much time on, and at times it wasn't very interesting for me personally

PiMuiscbox was aimed at people who didn't want / know how to customise a Raspbian system, maybe that's its niche and that should still be the focus. Problem is, there's always people who want something slightly different. I don't enjoy walking people through the hacks they need to customise pimusicbox because it's so inflexible.

MaStr, 2020-05-28:
So basically, I got a working setup for creating an image.
I'll dig further in my spare freetime :+1:

kingosticks, 2020-05-28:
Out of curiosity can I see how simple it is? I meant to take a look at the example you gave with octopi but it's still on my list of things to do.

MaStr, 2020-05-28:
https://github.com/MaStr/pimusicbox_CustomPiOS

It is an example, the path of the CustomPiOS in "custompios_path" is absolute to my homedirectory.

Prerequiste:
- docker
- docker-compose

You need to do the following:

- Create a work folder., i.e. "build"
- inside clone https://github.com/MaStr/pimusicbox_CustomPiOS  and https://github.com/guysoft/CustomPiOS
 - Adjust the path in custompios_path
 - move into the docker folder
 - run  ''docker-compose up -d"
 - got to the  pimusicbox_CustomPiOS / src folder
 - run  ''sudo docker exec -it  mypimusicbox build''

Currently the step after installing packages is failing. I can't see the error message, because the terminal is bogus after manually confirming a message from "samba" and a license confirm of "firmware-ipw2x00".

The things you do inside the image is configured in : src/modules/pimusicbox/start_chroot_script 

Configuration is done with shell variables, which are sourced from src/config

Modules caches for apt and other things are done automatigacally by the framework.

gvidasdj, 2020-06-10:
Is project still alive?

MaStr, 2020-06-14:
Explain "project"?

scls19fr, 2020-12-01:
An (other) interesting approach which could be achieved by people with docker-compose skills could be to create a distribution which could be installed using http://balena.io/cloud
Some examples https://hub.balena.io/

guysoft, 2022-03-17:
Hey, I just came across this thread. If you need help to figure out how to make CustomPiOS builds I can answer questions :)
It looks like @MaStr 's build is on the right track.