pimusicbox/pimusicbox #546: How to use Waveshare wm8960 external sound card?

kostasoft, 2020-06-24:
Can not install drivers by instructions from the official website:
https://github.com/waveshare/WM8960-Audio-HAT

kingosticks, 2020-06-24:
I've no idea, I've never heard of it. The kernel we use is a bit old, perhaps it's related to that. Sorry, can't help further.

kostasoft, 2020-06-24:
When you start the compilation it generates an error:
```
./install.sh: line 21: apt: command not found
Reading package lists... Done
Building dependency tree... Done
E: Unable to locate package raspberrypi-kernel-headers
E: Unable to locate package raspberrypi-kernel
Reading package lists... Done
Building dependency tree... Done
E: Unable to locate package dkms
E: Unable to locate package git
./install.sh: line 39: dkms: command not found
./install.sh: line 44: dkms: command not found
./install.sh: line 45: dkms: command not found
mkdir: cannot create directory `/etc/wm8960-soundcard': File exists
./install.sh: line 83: systemctl: command not found
./install.sh: line 84: systemctl: command not found
```
But on the freshly installed raspbian Lite, the compilation goes without errors.

kingosticks, 2020-06-24:
Change their install script to use `apt-get`. Also be aware pimusicbox is based on Debian Wheezy and there may be other incompatibilities in the install script. Good luck

JoebikerSMT, 2021-03-26:
Do you plan to support this sound card on any next release? @kostasoft were you able to make it run? If yes, would you please share how did you achieve it?