pimusicbox/pimusicbox #548: spotify connect unabe to change song.

RickyBurton, 2020-07-20:
hello, I can connect my Spotify to the pi music box and start a playlist, but it takes a while to connect, but if I want to go from bob Marley to sting I can not do it without a lot of hassle, I have the latest version of pi music box, but is there something I need to do with librespot to get it function.

KonecnyIgor, 2021-06-13:
I have got same problem. When on smartphone spotify app I have set for output device phone I can change my playlist for example Sting album. Then I switch to musicbox and I can only change next track on playlist. When I want to change for another playlist for example Sade then musicbox stop playing and start the sting somg what is on firsr playlist. To change playlist I must change output device to smartphone then change playlist and then switch to musicbox. It is not so userfriendly but only way what I know how to use spotify...

kingosticks, 2021-06-13:
There is the non-connect way to use Spotify. You open your web browser and visit the musicbox url and use that. So how you know another way to use Spotify. Yes, it might not be as user-friendly and it's not beyond the realm of anyone who's got this far.