pimusicbox/pimusicbox #552: SSH broken with newer clients

nerdoc, 2020-11-13:
I installed PiMusicBox and tried to ssh into it from an recent Manjaro/Arch ssh client (openssh 8.4p1-2), but this seems to be a problem too at Ubuntu 20.10...?:
When I try to connect, ssh moarns:

    Unable to negotiate with 192.168.1.<xxx> port 22: no matching key exchange method found. 
    Their offer: diffie-hellman-group1-sha1,diffie-hellman-group14-sha1

So I suppose that your ssh server uses just these old algorhythms, and newer clients even ignore these because they are too old and insecure.
Could you add support for newer algorhythms? Because else it is not possible to connect via ssh with any newer Linux distro...

nerdoc, 2020-11-13:
It is possible to workaround this in the client using

    ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 musicbox.local

At least you can connect to it.

But this is not the fix. Please update the algorithms...

tsopanid, 2020-11-20:
for me even this doesn't work after fresh install.
`ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 pi@musicbox.local`  

neither with default pw 'raspberry' nor with the one i create on the web-interface;
> Permissions denied, please try again.

tsopanid, 2020-11-25:
Tried again today and could log in.
Didn't know i could login as root, thought it was raspains default user 'pi'.
So the workaround works :)

nerdoc, 2020-11-26:
But it's no fix ;-)

tsopanid, 2020-11-26:
True!