pimusicbox/pimusicbox #557: I can't get this to work on the Raspberry Pi Zero W 2?

cgchris99, 2021-12-17:
Anyway to get this to run on the PI Zero W 2?

nepozs, 2021-12-25:
Try this (copying .dtb file):
https://forums.raspberrypi.com/viewtopic.php?t=322744&start=25
untested but may work.

jarethmt, 2022-02-18:
Just tested this out, can confirm that it does NOT work unfortunately. Anyone else have any ideas? Very unfortunate that this project seems dead and doesn't support a lot of the modern hardware :(

kingosticks, 2022-02-18:
Yes, it is dead. Hardware support stopped at 3B. If you want to discuss workarounds I think the [forum](https://discourse.mopidy.com/t/pimusicbox-pizero2w-support/5258/5) might be more appropriate, although I don't know of a working solution there yet either.